libremesh/lime-sdk #2: more examples or brunch to get use of lede-trunk

FreifunkUFO, 2017-04-04:
get use of lede-trunk (instead of official release) is nicer for testing our testing firmware testings. (and buildin g testing firmwares for further tests

FreifunkUFO, 2017-04-04:
i know: building the right imagebuilder/sdk is another task. 
but, do i have to set both imagebuilder AND sdk via "-i" option within the script?

so, more examples for building testing firmwares would be nice. 
otherwise all of us will build firmware a bit different, by own workarounds..

p4u, 2017-04-05:
The idea behind lime-sdk is to use the official LEDE precompiled SDK and IB. So the generated firmware will be 100% compatible with official LEDE repositories.

The cooker script currently downloads the LEDE 17.01.0 release SDK and IB to compile and generate libremesh. But the idea is to keep different branches into lime-sdk, one for stable release and another for the current release snapshot. So I'll add soon a branch for working with LEDE 17.01 development snapshot.

p4u, 2017-04-05:
And yes, you are right that the documentation is still a pending task. I'll work on this the next days before announcing this tool as the official tool for building libremesh firmware.

p4u, 2017-04-05:
I just added a new branch named "develop" which points to current LEDE 17.01 snapshot SDK/IB

https://github.com/libremesh/lime-sdk/tree/develop

So if you want to compile the last LEDE source, execute "git checkout develop" on a new (vanilla) lime-sdk.

Let me know if it works for you.

FreifunkUFO, 2017-04-05:
thanks for the develop brunch.
its working for me, output is http://gadow.freifunk.net:8004/srv2/libremesh/develop-dev-brunch/lime-sdk/output/ar71xx/generic/Default/lime_default/

p4u, 2017-04-05:
I don't get why the "Default" directory, that does not seems correct to me. Can you post the command line used?

FreifunkUFO, 2017-04-05:
i was using:

git clone https://github.com/libremesh/lime-sdk.git

cd lime-sdk

git checkout develop

    ./cooker -d ar71xx/generic                        # download SDK and IB 

    ./cooker -f                                       # download and prepare feeds

    ./cooker -b ar71xx/generic                        # build the SDK and link it to IB

    ./cooker -c ar71xx/generic --flavor=lime_default  # cook the firmware

FreifunkUFO, 2017-04-05:
is there a possibility (or example) to build libremesh firmware for devices with 4MB-flash? otherwise we have to test with 8mb-flash-devices.

p4u, 2017-04-05:
There is not way for the moment... As we use LEDE SDK we cannot modify Kernel, so we need to reduce the space from somewhere else. We have to figure out how to do it.

FreifunkUFO, 2017-04-06:
so micha is testing a nano-profile with this lime-sdk right now..

FreifunkUFO, 2017-04-10:
4mb devices now with flavour nano working.. 
before we can have an example for README we have to do this:

musst du nen pr aufmachen, den vorhin verlinkten commit cherrypicken auf den develop branch

FreifunkUFO, 2017-04-26:
nano is now called zero :-o
(lime_nano heißt jetzt lime_zero)

FreifunkUFO, 2017-04-27:
im closing this ticket, it doesnt make sense without feedback.
- these 4 commands above dont create libremesh-firmware, but a normal lede-firmware
- all people i asked dont have a problem with that README at all.
- the develop brunch seems to be obsolete
- compiling libremesh with lede-trunk seems to be out-of-scope (so i have to wait some months for testing my tplinks 850v2)

p4u, 2017-04-27:
You are right, I've put all the efforts on release 17.01. But as LEDE will also release every few months (the current one is from some days ago 17.01.1) lime-sdk release branch will be also updated.

I will put more efforts on develop as soon as I can.

Right now it is possible to create mini (with web interface) and zero (without web interface) for 4MB routers.

I hope we can have your feedback.

FreifunkUFO, 2017-05-03:
i did a flavor "lime-micro" to fit on 4mb devices and to have batctl.
To have libremesh only on lede-releases is nice for stable releases.. but, when having lede- or packages-bugfixes we will be forced to wait a few months to check / continuing our development.

darkdrgn2k, 2017-05-27:
LEDE Trunk from 850v2 using stock BUILDROOT and lime feeds

#Clone BuildRoot
git clone https://github.com/lede-project/source

#Add Feed
cd source
cp feeds.conf.default feeds.conf
echo "src-git lime https://github.com/libremesh/lime-packages.git" >> feeds.conf

#Update and Install Feed
./scripts/feeds update -a
./scripts/feeds install -a

#config
make menuconfig
**ill post my config file later**

#make
make

FreifunkUFO, 2017-05-27:
now i will try:

i was using:

git clone https://github.com/freifunk-leipzig/libremesh-lime-sdk.git

cd lime-sdk

git checkout ledetrunk

    ./cooker -d ar71xx/generic                        # download SDK and IB 

    ./cooker -f                                       # download and prepare feeds

    ./cooker -b ar71xx/generic                        # build the SDK and link it to IB

    ./cooker -c ar71xx/generic --flavor=lime_default  # cook the firmware

FreifunkUFO, 2017-05-27:
the -b command fails with:

 make[3] -C /srv2/libremesh/20170528/libremesh-lime-sdk/feeds/base/package/network/utils/iptables compile
 make[3] -C /srv2/libremesh/20170528/libremesh-lime-sdk/feeds/base/package/kernel/mac80211 compile
make -r world: build failed. Please re-run make with -j1 V=s to see what's going on
/srv2/libremesh/20170528/libremesh-lime-sdk/17.01.1/ar71xx/generic/sdk/include/toplevel.mk:190: recipe for target 'world' failed
make: *** [world] Error 1
make: Leaving directory '/srv2/libremesh/20170528/libremesh-lime-sdk/17.01.1/ar71xx/generic/sdk'
-> Error compiling SDK

p4u, 2017-05-28:
If you need to check a message error, you should execute the cooker with V=s and J=1

`V=s J=1 ./cooker -b ar71xx/generic`

This is the current error when compiling LEDE SDK _trunk_.

> rm -f /home/repos/lime-sdk/lede/ar71xx/generic/sdk/staging_dir/target-mips_24kc_musl/stamp/.mac80211_installed
> cmp /home/repos/lime-sdk/lede/ar71xx/generic/sdk/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/compat-wireless-2017-01-31/include/linux/ath9k_platform.h /home/repos/lime-sdk/lede/ar71xx/generic/sdk/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.4.70/include/linux/ath9k_platform.h
> /home/repos/lime-sdk/lede/ar71xx/generic/sdk/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/compat-wireless-2017-01-31/include/linux/ath9k_platform.h /home/repos/lime-sdk/lede/ar71xx/generic/sdk/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.4.70/include/linux/ath9k_platform.h differ: char 1483, line 52
> make[3]: *** [Makefile:1780: /home/repos/lime-sdk/lede/ar71xx/generic/sdk/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/compat-wireless-2017-01-31/.configured_01f75d3526bf412a534c78e54e99897f] Error 1

So ath9k compilation is crashing, I think you should ask LEDE.

FreifunkUFO, 2017-05-28:
V and J before cooker command? there is no documentation about that :-o
i was trying V=99 behind cooker command, with no impact.. and i thought without j=x there is j=1. j has to be capitalized?

but i found another mistake in my modified cooker script and was fixing it already. (download-url)
next problem (see above comment) is f.i. to fix existence of files in /rom/usr/lib/opkg/. might be another ticket (or better check lede about that problem?).
and next problem could be to shrink firmware once more to fit on 4mb devices?

FreifunkUFO, 2017-05-28:
yesterday @darkdrgn2k tried it with disabling a nat-package, and stripped busybox a bit (inside "make menuconfig", without lime-sdk, and without documentation). but still too big firmware image to have jffs

FreifunkUFO, 2017-05-30:
here is a ticket for the bug @darkdrgn2k found:

imagebuilder doesnt delete /usr/lib/opkg anymore

https://bugs.lede-project.org/index.php?do=details&task_id=817

FreifunkUFO, 2017-05-30:
we might also do not need /ar71xx/generic/ib/build_dir/target-mips_24kc_musl/root-ar71xx/etc/board.d/01_leds

37k

FreifunkUFO, 2017-05-31:
i heared @NeoRaider is changing things inside imagebuilder atm, maybe to fit it to gluon.

FreifunkUFO, 2017-10-31:
i dont understand these settings there, it seems to be a wild mix. https://github.com/libremesh/lime-sdk/compare/develop but its not lede-trunk in there. maybe its also affected from that "frozen wireless-compat"?

ilario, 2017-10-31:
On #35 I described a ugly workaround for compiling with current develop branch.

I don't think that @p4u wants to use develop for compiling on LEDE trunk, but on 17.01, is this correct?
Anyway @FreifunkUFO can you make a PR against develop with how should be the feeds.conf.default for compiling on LEDE trunk?

FreifunkUFO, 2017-10-31:
@ilario no. i dont tried it within the last months.
but you might only change "lede-17.01" to "master" sererval times in the feeds.conf.default

p4u, 2017-11-01:
The current develop branch of lime-sdk should compile LEDE 17.01 in-development version. On feeds.conf LEDE source points to 

https://git.lede-project.org/source.git;lede-17.01

Which is the in-development (previously known as trunk) branch.

darkdrgn2k, 2018-01-21:
So im jumping back on the band wagon here :)

so new cooker can build wa850re
```./cooker -c ar71xx/tiny --profile=tl-wa850re-v2 --flavor=lime_zero```

but image is "too large"

```
4336+1 records out
2220114 bytes (2.2 MB, 2.1 MiB) copied, 0.0165327 s, 134 MB/s
os-image partition too big (more than 1376256 bytes): Success
cp: cannot stat '/home/user/libremesh/lime-sdk/lede/ar71xx/tiny/ib/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tmp/openwrt-lime-zero-ar71xx-tiny-tl-wa850re-v2-squashfs-sysupgrade.bin': No such file or directory
4336+1 records in
4336+1 records out
2220114 bytes (2.2 MB, 2.1 MiB) copied, 0.0146831 s, 151 MB/s
os-image partition too big (more than 1376256 bytes): Success
cp: cannot stat '/home/user/libremesh/lime-sdk/lede/ar71xx/tiny/ib/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tmp/openwrt-lime-zero-ar71xx-tiny-tl-wa850re-v2-squashfs-factory.bin': No such file or directory
```

working on a new flavoud
```
lime_minimal="librt lime-hwd-ground-routing libc kmod-ebtables lime-proto-bmx6 smonit bmx6-auto-gw-mode libpthread lime-system lime-proto-batadv lime-hwd-openwrt-wan dnsmasq-lease-share lime-proto-anygw lime-proto-wan dnsmasq-distributed-hosts -lime-proto-wan -kmod-pppoe -kmod-pppox -kmod-ppp -opkg -wpad-mini hostapd-mini -kmod-usb-core -kmod-usb-ledtrig-usbport -kmod-usb2 -ppp -dnsmasq -ppp-mod-pppoe -6relayd -odhcp6c -odhcpd -iptables -ip6tables -luci-app-lime-location -luci-lib-nixio -libremap-agent -luci-lib-libremap -luci-lib-libremap-bmx6 -luci-lib-libremap-location -luci-lib-libremap-system -luci-lib-libremap-wireless -lime-map-agent -swconfig -ebtables -lime-docs-minimal"
```
still way out of line for the min size

However if i can figure out how to set CLEAN_IPKG for the sdk install it may save quite a bit of space.

FreifunkUFO, 2018-02-19:
ping @dangowrt, you are testing with snapshot at the moment..  
what is needed to change? are you trying with openwrt?

darkdrgn2k, 2018-02-19:
If that was @ me then yes i am

I ran into a snag and bricked one of my nodes.  (It may have been that i flashed the wrong image or the build was wrong)

Anyway Warning: to unbrick it you have to crack the case and the device will never look the same again (sonic welding)

Latest config i tried to have  enough free space to add overlay
https://pastebin.com/4GwWCywJ

Its getting harder and harder to squeeze everything in!

dangowrt, 2018-02-19:
i built with lime-sdk, but don't have a chance to test on real
hardware (in our crowded kitchen)... also now that i'm home i
got less than 20kb/s which is insufficient even for just downloading
the package lists of the IB (which need to run through every time...)
changes were rather trivial, some changes to options.conf and
feeds.conf.default to use snapshot as version and then it run through.

```
diff --git a/options.conf b/options.conf
index 99187df..c166762 100644
--- a/options.conf
+++ b/options.conf
@@ -1,5 +1,5 @@
-release=17.01.4
-base_url=https://downloads.lede-project.org/releases/$release/targets/
+release=snapshot
+base_url=https://downloads.lede-project.org/snapshots/targets
 communities_git=https://github.com/libremesh/network-profiles.git
 communities_dir=communities
 tmp_dir=tmp
@@ -17,4 +17,4 @@ targets_list=targets.list
 downloads_dir=dl
 make_j=1
 bin_output=output
-brand_name=lede
+brand_name=openwrt
```

In feeds.conf.default simply remove the references to lede-17.01 and
use 'master' instead, also replace '17.06' with 'develop' for the
libremesh feed.

darkdrgn2k, 2018-02-19:
Ah mine is for the tl-wa850re-v2 to try to save space.. never mind.

FreifunkUFO, 2018-02-20:
@dangowrt please reboot 206-41 (or better that ubnt behind it) to get more speed

aparcar, 2018-05-30:
Closed due to no reponse

@FreifunkUFO please reopen if still required.

closed 2018-05-30 by aparcar