libremesh/lime-sdk #3: The update breaks if feeds in feeds.conf references a tag instead of a branch

nicopace, 2017-05-05:
This happens with:
```
https://github.com/lede-project/source.git;v17.01.1 
```
because v17.01.1 is a tag,
and does not happen with
```
src-git libremesh https://github.com/libremesh/lime-packages.git;develop
```
That is a branch.

The issue comes because for each feeds.conf line you are doing git pull without checking if it is a tag or branch here: https://github.com/libremesh/lime-sdk/blob/master/cooker#L213
You can check the available tags by using ```git tag -l```

p4u, 2017-05-05:
Hi.
What you mean by break exactly? Can you attach some output?

If "git pull" is executed over a tag it does just nothing, so IMO it is not breaking anything. The only thing is that git complains... It might be controlled but does not affect in any way to the whole cooking process.

nicopace, 2017-05-05:
Sorry about the alarming message... we were having issues and thought it was because of that error.
It does nothing bad, but it would be a good aesthetic improvement to just check if it is a tag don't do the pull, because it is not needed, and because it throws a visual error.

p4u, 2017-05-05:
Agree.
I'll do it when I have time but it is not a priority in my tasks list. If you want to send a PR, it will be welcome :)