libremesh/lime-sdk #4: x86/generic: Cannot install package lime-full

gmarcos87, 2017-05-11:
I am trying to cook a generic x86 image of Libremesh to run in VirtualBox. I execute:
```
./cooker -c x86/generic --profile=Generic --flavor=lime_default
```
And I get this error as a result:
```
[Long list of "Configuring ...." ]
Collected errors:
 * opkg_install_cmd: Cannot install package lime-full.
Makefile:140: recipe for target 'package_install' failed
make[2]: *** [package_install] Error 255
make[2]: Leaving directory '/home/okupa/libremesh/lime-sdk/17.01.0/x86/generic/ib'
Makefile:110: recipe for target '_call_image' failed
make[1]: *** [_call_image] Error 2
make[1]: Leaving directory '/home/okupa/libremesh/lime-sdk/17.01.0/x86/generic/ib'
Makefile:196: fallo en las instrucciones para el objetivo 'image'
make: *** [image] Error 2
make: se sale del directorio '/home/okupa/libremesh/lime-sdk/17.01.0/x86/generic/ib'

```

p4u, 2017-05-11:
It means that lime packages have not been compiled...

That should be done automatically, but try execute first:
./cooker -b x86/generic

And check if there is any error.

On 11/05/17 15:25, Marcos Gutierrez wrote:
> I am trying to cook a generic x86 image of Libremesh to run in VirtualBox. I execute:
> ```
> ./cooker -c x86/generic --profile=Generic --flavor=lime_default
> ```
> And I get this error as a result:
> ``` [...]

-- 
./p4u

gmarcos87, 2017-05-11:
I tried, same error. This is the output of ./cooker -b x86/generic
```
-> Building 17.01.0/x86/generic/sdk
Updating feed 'base' from '/home/okupa/libremesh/lime-sdk/feeds/base' ...
Create index file './feeds/base.index' 
Updating feed 'packages' from '/home/okupa/libremesh/lime-sdk/feeds/packages' ...
Create index file './feeds/packages.index' 
Updating feed 'luci' from '/home/okupa/libremesh/lime-sdk/feeds/luci' ...
Create index file './feeds/luci.index' 
Updating feed 'routing' from '/home/okupa/libremesh/lime-sdk/feeds/routing' ...
Create index file './feeds/routing.index' 
Updating feed 'telephony' from '/home/okupa/libremesh/lime-sdk/feeds/telephony' ...
Create index file './feeds/telephony.index' 
Updating feed 'libremesh' from '/home/okupa/libremesh/lime-sdk/feeds/libremesh' ...
Create index file './feeds/libremesh.index' 
Updating feed 'libremap' from '/home/okupa/libremesh/lime-sdk/feeds/libremap' ...
Create index file './feeds/libremap.index' 
Updating feed 'limeui' from '/home/okupa/libremesh/lime-sdk/feeds/limeui' ...
Create index file './feeds/limeui.index' 
Installing all packages from feed libremesh.
Installing all packages from feed libremap.
Installing all packages from feed limeui.
WARNING: No feed for package 'libustream-openssl' found, maybe it's already part of the standard packages?
make: se entra en el directorio '/home/okupa/libremesh/lime-sdk/17.01.0/x86/generic/sdk'
make[1]: Entering directory '/home/okupa/libremesh/lime-sdk/17.01.0/x86/generic/sdk'
make[1]: Leaving directory '/home/okupa/libremesh/lime-sdk/17.01.0/x86/generic/sdk'
tmp/.config-package.in:36:warning: ignoring type redefinition of 'PACKAGE_libc' from 'boolean' to 'tristate'
tmp/.config-package.in:64:warning: ignoring type redefinition of 'PACKAGE_libgcc' from 'boolean' to 'tristate'
tmp/.config-package.in:121:warning: ignoring type redefinition of 'PACKAGE_libpthread' from 'boolean' to 'tristate'
tmp/.config-package.in:149:warning: ignoring type redefinition of 'PACKAGE_librt' from 'boolean' to 'tristate'
#
# configuration written to .config
#
make: se sale del directorio '/home/okupa/libremesh/lime-sdk/17.01.0/x86/generic/sdk'
make: se entra en el directorio '/home/okupa/libremesh/lime-sdk/17.01.0/x86/generic/sdk'
make[1]: Entering directory '/home/okupa/libremesh/lime-sdk/17.01.0/x86/generic/sdk'
make[2]: Entering directory '/home/okupa/libremesh/lime-sdk/17.01.0/x86/generic/sdk'
make[2]: Leaving directory '/home/okupa/libremesh/lime-sdk/17.01.0/x86/generic/sdk'
make[2]: Entering directory '/home/okupa/libremesh/lime-sdk/17.01.0/x86/generic/sdk'
make[2]: Leaving directory '/home/okupa/libremesh/lime-sdk/17.01.0/x86/generic/sdk'
make[1]: Leaving directory '/home/okupa/libremesh/lime-sdk/17.01.0/x86/generic/sdk'
tmp/.config-package.in:36:warning: ignoring type redefinition of 'PACKAGE_libc' from 'boolean' to 'tristate'
tmp/.config-package.in:64:warning: ignoring type redefinition of 'PACKAGE_libgcc' from 'boolean' to 'tristate'
tmp/.config-package.in:121:warning: ignoring type redefinition of 'PACKAGE_libpthread' from 'boolean' to 'tristate'
tmp/.config-package.in:149:warning: ignoring type redefinition of 'PACKAGE_librt' from 'boolean' to 'tristate'
#
# configuration written to .config
#
 make[1] world
 make[2] package/compile
 make[3] -C package/toolchain compile
 make[3] -C package/linux compile
 make[2] package/index
```

p4u, 2017-05-11:
Something is broken in your lime-sdk. It looks like some feeds are
missing. Maybe the "git clone" failed when downloading the feeds.

Can you try to start from scratch and execute step-by-step ? (example
for ar71xx/generic)

./cooker -d ar71xx/generic
./cooker -f
./cooker -b ar71xx/genericlink it to IB
./cooker -c ar71xx/generic --flavor=lime_default

On 11/05/17 15:49, Marcos Gutierrez wrote:
> I tried, same error. This is the output of ./cooker -b x86/generic
> ```
> -> Building 17.01.0/x86/generic/sdk
> Updating feed 'base' from '/home/okupa/libremesh/lime-sdk/feeds/base' ...
> Create index file [...]

-- 
./p4u

gmarcos87, 2017-06-02:
The problem occurs when one of the packets fails to download. In intermittent connections or of poor quality it is almost impossible to finish downloading all the packages without at least one fails, and the solution is to try again to download all new ones. We should be able to identify which packets failed and be able to remove those downloads. The "solution" was to copy with a pendrive the folder of another user.

ilario, 2017-06-02:
@gmarcos87 in case of a failed download usually is enough to issue again the same command. Many packets have MD5 check so they will not download broken.

gmarcos87, 2017-06-03:
How to reproduce:

- Clone lime-sdk
- Run ```./cooker -c x86/generic --profile=Generic```
- Interrupt the Internet connection when cloning feeds and wait for download timeout
- Compilation will fail as expected
- If re-executing ```./cooker -c x86/generic --profile=Generic``` with the active Internet connection, the missing feeds are not resumed or attempted.

ilario, 2017-06-04:
Ok, this is very interesting, I would consider this as a lime-sdk bug. Can you update the issue title?

p4u, 2017-06-05:
Will be something like "Check correct download of feeds"