libremesh/lime-sdk #6: args for custom packages and custom flavor name

aparcar, 2017-05-16:
It could be useful to set to be installed packages via command line as well define a custom flavor name.

    ./cooker -c ar71xx/generic --profile=tl-wdr3600-v1 --packages "luci-ssl -kmod-usb-core" --flavor-name=gateway-foobar

the created image would be created at `lime-sdk/output/ar71xx/generic/tl-wdr3600-v1/gateway-foobar/`

p4u, 2017-05-16:
Right now you can use cooker like this to achieve what you are proposing (if I understood correctly).

> export PKG="luci-ssl -kmod-usb-core"
> ./cooker -c ar71xx/generic --profile=tl-wdr3600-v1 --flavor=gateway-foobar

If the flavor does not exist in flavors.conf then only the default LEDE set of packages are selected. The PKG system variable is used to specify extra packages.

However, using `--packages` instead (or in addition to) a system variable might be more explicit and easy to understand for the user. If you want to include it, I'd review and accept a pull request.

p4u, 2017-07-10:
Added on commit 5c27eb26712eb641fcddef7662bad0c33c2ad9f7

closed 2017-07-10 by aparcar