libremesh/lime-sdk #11: Build a single package

nicopace, 2017-07-09:
As a package developer I need to be able to run the compilation step of my package many times and as fast as possible.
Right now, with the lime-sdk I need to wait for all the image to build... this can take ~1 minute, while building the package directly using openwrt-sdk takes 2 seconds.
Would be great to bypass the building of all but one package, and then optionally be able to create an image with that new packge.
Some conversation about this topic here: https://github.com/libremesh/lime-sdk/commit/f53d7386430b64eb7529562b0471c707412666dd#commitcomment-22994952

p4u, 2017-07-10:
Added on: 
https://github.com/libremesh/lime-sdk/commit/127b9cdc10dc8076cd01504e2332031646bd3429

nicopace, 2017-07-11:
It is telling me this:
```
nicopace@nicopc:~/projects/redlibre/altermundi/librestack/lime-sdk$ J=1 V=s ./cooker -b ar71xx/generic --package=fft-eval
...
/home/nicopace/projects/redlibre/altermundi/librestack/lime-sdk/lede/ar71xx/generic/sdk/include/toplevel.mk:200: recipe for target 'package/fft-eval/install' failed
make: *** [package/fft-eval/install] Error 2
make: Leaving directory '/home/nicopace/projects/redlibre/altermundi/librestack/lime-sdk/lede/ar71xx/generic/sdk'
-> Build error, see full output using V=s J=1 ./cooker <options>
```
And I've run it with V=s and J=1

ilario, 2017-08-29:
I think this is solved, is it right?

nicopace, 2017-08-29:
yes... thanks

closed 2017-08-29 by nicopace