libremesh/lime-sdk #19: consolidation of package

dangowrt, 2017-08-01:
Currently packages are passed to the ImageBuilder by simply appending the string coming from flavours.conf with the PKG environment variable. This is insufficient in case of the user wanting to install a package via PKG variable which is masked by a `-` character in front of the package. In such a case (such as `firewall`) the package cannot be installed via the PKG variable at all.

p4u, 2017-08-03:
Is this not working for you?

`./cooker -c ar71xx/generic --profile=tl-wdr4300-v1 --flavor=lime_default --extra-pkg="-firewall luci-app-3g"`

p4u, 2017-08-03:
Ah, ok! You mean that packages which are market with minus on the flavor cannot be installed by the user.

So we need to implement a funcion on cooker to join the package lists and add preference to the user choice.

dangowrt, 2017-08-03:
Precisely. You got it.

p4u, 2017-09-15:
implemented on commit https://github.com/libremesh/lime-sdk/commit/75ce2b3b099da646d2dbf1705769fb94a59c2131

So by order of preference:

 1. User defined packages on --extra-pkg
 2. Packages on global PKG var
 3. Community profile packages
 4. Flavor packages

closed 2017-09-15 by p4u