libremesh/lime-sdk #28: Image build failing.  Cannot install package lime-full

develroo, 2017-10-10:
Hi,

Pulled the git. Followed the instructions and tried to built and image but got the following error.

```
Collected errors:
 * opkg_install_cmd: Cannot install package lime-full.
Makefile:140: recipe for target 'package_install' failed
make[2]: *** [package_install] Error 255
make[2]: Leaving directory '/usr/src/git/lime-sdk/17.01.2/ar71xx/generic/ib'
Makefile:110: recipe for target '_call_image' failed
make[1]: *** [_call_image] Error 2
make[1]: Leaving directory '/usr/src/git/lime-sdk/17.01.2/ar71xx/generic/ib'
Makefile:196: recipe for target 'image' failed
make: *** [image] Error 2
make: Leaving directory '/usr/src/git/lime-sdk/17.01.2/ar71xx/generic/ib'

-> Firmware not cooked, something wrong happened on the ImageBuilder compilation process

```

even running with 

```J=1 V=s ./cooker -c ar71xx/generic  --profile=tl-wr940n-v4 --flavor=lime_default ```

Not sure where to go now?

ilario, 2017-10-10:
There should be more errors earlier in the log, can you post them also?

develroo, 2017-10-10:
Hmmm looks like an error building openssl. Seems there are whole load of undefined functions. 
```
openssl.o: In function `main':
openssl.c:(.text.startup.main+0xf4): undefined reference to `CRYPTO_set_mem_debug_functions'
openssl.c:(.text.startup.main+0xf8): undefined reference to `CRYPTO_dbg_get_options'
openssl.c:(.text.startup.main+0xfc): undefined reference to `CRYPTO_dbg_set_options'
openssl.c:(.text.startup.main+0x100): undefined reference to `CRYPTO_dbg_free'
openssl.c:(.text.startup.main+0x104): undefined reference to `CRYPTO_dbg_realloc'
openssl.c:(.text.startup.main+0x108): undefined reference to `CRYPTO_dbg_malloc'
openssl.c:(.text.startup.main+0x10c): undefined reference to `CRYPTO_set_mem_debug_functions'
openssl.c:(.text.startup.main+0x118): undefined reference to `CRYPTO_set_mem_debug_options'
openssl.c:(.text.startup.main+0x11c): undefined reference to `CRYPTO_set_mem_debug_options'
...
```
I have uploaded the output [here](https://dropfile.to/7FGmHU4).

hope that helps.

p4u, 2017-10-11:
> Hmmm looks like an error building openssl. Seems there are whole load
> of undefined functions.
>
> |openssl.o: In function `main': openssl.c:(.text.startup.main+0xf4):
> undefined reference to [...]

Not the first time I see this error but I'm not sure which is the real
cause. However it is something related to the host machine, not to the
lede/libremesh SDK.
Which distro do you use? Did you install all the dependencies?
https://lede-project.org/docs/guide-developer/install-buildsystem

ilario, 2017-10-11:
Looks like what you're missing on your machine is `libssl-dev`

develroo, 2017-10-11:
Nope. It is not that. FTR I am using Debian/Testing

```
dpkg-query -s  libssl-dev
Package: libssl-dev
Status: install ok installed
Priority: optional
Section: libdevel
Installed-Size: 6897
Maintainer: Debian OpenSSL Team <pkg-openssl-devel@lists.alioth.debian.org>
Architecture: amd64
Multi-Arch: same
Source: openssl
Version: 1.1.0f-5
Depends: libssl1.1 (= 1.1.0f-5)
Recommends: libssl-doc
Conflicts: libssl1.0-dev
Description: Secure Sockets Layer toolkit - development files
 This package is part of the OpenSSL project's implementation of the SSL
 and TLS cryptographic protocols for secure communication over the
 Internet.
 .
 It contains development libraries, header files, and manpages for libssl
 and libcrypto.
Homepage: https://www.openssl.org/
```

develroo, 2017-10-11:
I think I found the problem.

Looking at the Makefile in lime-sdk/feeds/base/package/libs/openssl

```
PKG_NAME:=openssl
PKG_BASE:=1.0.2
PKG_BUGFIX:=k
PKG_VERSION:=$(PKG_BASE)$(PKG_BUGFIX)
PKG_RELEASE:=1
PKG_USE_MIPS16:=0

PKG_BUILD_PARALLEL:=0

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=http://www.openssl.org/source/ \
```

points to openssl-1.0.2**k**

k does not exist. The latest patch version is l.

I will edit the Makefile and see if that fixes it.

develroo, 2017-10-11:
Yup. That seemed to fix it. 

N.B. You also need to include the sha256sum value into the Makefile, which sit alongside the tar.gz file.

develroo, 2017-10-11:
Hmmm. I thought I was going to end up with a sysupgrade.bin file instead I end up with others?

```
-rw-r--r-- 1 root root    4661 Oct 11 15:17 lede-17.01.2-lime-default-ar71xx-generic-device-tl-wr940n-v4.manifest
-rw-r--r-- 1 root root 3538944 Oct 11 15:17 lede-17.01.2-lime-default-ar71xx-generic-root.squashfs
-rw-r--r-- 1 root root 1257014 Oct 11 15:17 lede-17.01.2-lime-default-ar71xx-generic-uImage-lzma.bin
-rwxr-xr-x 1 root root 4018388 Oct 11 15:17 lede-17.01.2-lime-default-ar71xx-generic-vmlinux.bin
-rwxr-xr-x 1 root root 4023512 Oct 11 15:17 lede-17.01.2-lime-default-ar71xx-generic-vmlinux.elf
-rw-r--r-- 1 root root 1310720 Oct 11 15:17 lede-17.01.2-lime-default-ar71xx-generic-vmlinux.lzma
-rwxr-xr-x 1 root root 1329012 Oct 11 15:17 lede-17.01.2-lime-default-ar71xx-generic-vmlinux-lzma.elf
-rw-r--r-- 1 root root     862 Oct 11 15:17 sha256sums
```
What am I missing?

ilario, 2017-10-11:
The file with the **k** does exist
https://www.openssl.org/source/openssl-1.0.2k.tar.gz
but maybe they fixed the error you hit in **l** revision?
I don't know, can someone else on Debian testing reproduce this? The compilation was ok for me on Arch.

You didn't get the image compiled 'cause you're using the lime_default flavor (for routers with more than 4 MB of flash memory) for a router with 4 MB flash memory.
lime-sdk doesn't warn you yet, this has been reported in #21 
For obtaining the images you need to use the lime_mini (without package manager, this is the recommended one) or lime_zero (without web interface) flavors.

ilario, 2017-10-19:
@develroo did you manage to compile the image? Shall I close this?

develroo, 2017-10-20:
Yes I did build it in the end but there really ought to be a warning of some sort if  you are building to a target with or less than 4Mb that you should not build the lime-full image  only lime_mini or lime_zero ones.

Thanks for the help. Got there in the end.

ilario, 2017-10-20:
> there really ought to be a warning of some sort if you are building to a target with or less than 4Mb that you should not build the lime-full image only lime_mini or lime_zero ones.

Yep, I agree, this would be partially solved in #21. A way for detecting routers flash size is also interesting for jefe/chef and has been requested to LEDE people: https://lists.libremesh.org/pipermail/lime-dev/2017-October/000994.html

closed 2017-10-20 by ilario