libremesh/lime-sdk #30: files from community profile not in files_dir folder

bvujnovac, 2017-10-17:
#cp -rf ${myfiles}/* ${ib_files}/ 2>/dev/null
 cp -rf ${ib_files}/* ${myfiles}/ 2>/dev/null

I had a problem with files from community not being in files_dir.
the first, or commented line is original code that was in the cooker file I simply switched places of files_dir and tmp_dir variables and now it works(as in code from the second line)

p4u, 2017-10-18:
Do you use the last version of lime-sdk (master branch) ?

What you say do not have much sense fore me... $ib_files is the directory used by imagebuilder to include the files when cooking -c, by default at lime-sdk/tmp/files/

So the content from $myfiles (which is the directory lime-sdk/files) is copied to $ib_files to be copied on the resulting firmware when Imagebuilder is executed.

p4u, 2017-12-15:
Should be solved now.

Thanks for reporting.

aparcar, 2018-05-30:
This was fixed a while ago

closed 2018-05-30 by aparcar