libremesh/lime-sdk #31: --update-feeds by default?

ilario, 2017-10-19:
In my opinion cooker should update the feeds (`--update-feeds`) by default, and having a `--no-update-feeds` required for avoiding this.
This would limit situations where users compile broken formware like [here](https://lists.libremesh.org/pipermail/lime-users/2017-October/000911.html).

aparcar, 2017-10-29:
Sound good.

p4u, 2017-11-19:
Agree

nicopace, 2018-06-20:
I would say it would be relevant to tell the user that there are updates that can be done, but not enforce by default to update.
It is an undesired sideeffect if you want to have reproducible builds, as each time you run build you may get different packages without the parameters changing in your commandline.
I would propose for the system to do a non-destructive action of checking if there were changes that could be applied by fetching but not merging the remote changes.
It is as easy as doing fetch on each git repo, and showing how many commits are from the previous to the current situation.

G10h4ck, 2018-10-22:
The reproducibility of the builds is guaranted for releases anyway because of tags used for repository feeds, but having to specify manually to update the feeds is quite antipattern in almost any case, so I would merge the related pullrequest

nicopace, 2018-10-22:
Like the approach... will review the code now.

nicopace, 2018-10-22:
jej, thought it was a PR... like the approach proposed by @G10h4ck

ilario, 2018-10-23:
Fixed by #70

closed 2018-10-23 by ilario