libremesh/lime-sdk #32: promote https://imagebuilder.libremesh.org/

aparcar, 2017-10-23:
The service slowly becomes more stable so some people might like to use the online editor instead of setting up their own builder

* https://imagebuilder.libremesh.org/
* https://ib.libremesh.org/

FreifunkUFO, 2017-10-29:
would ne nice, to finish some things first. the name is discussed at lime-dev malingliste atm. some parameters are changing atm and some final tests have to be done before giviing it more people for usage.

ilario, 2017-10-29:
Ok :) let's pause #34 (anyway we already linked that in lime-package readme). For the name, is also the subdomain imagebuilder.libremesh.org going to change or just the "jefe" name?

aparcar, 2017-11-04:
@ilario we decided to use chef.libremesh.org

ilario, 2017-11-06:
@aparcar great, going to modify the PR #34

ilario, 2017-11-06:
Wait, which will be the best URL? `https://chef.libremesh.org/imagebuilder` or `https://chef.libremesh.org`?

ilario, 2017-11-11:
It's the second one.

commit b7412646 (2017-11-11): Merge pull request #34 from ilario/issue/32

Added online imagebuilder to readme, fix #32
closed 2017-11-11 by ilario