libremesh/lime-sdk #35: Compile error

AndroidCloned, 2017-10-30:
I cannot successfully compile Target profiles for the ar71xx/generic target=Archer-C59-V1 or Ramips/7620 target=EX3700-3800 

I have tried several environment variables and a clean git clone using the Development branch as both of these profiles are not found in the master branch. Strangely enough I have a working archer c59 firwmare image that UFO built in April which is partially working without the latest patch that resolves wan port and switch port 1 ( http://gadow.freifunk.net:8004/srv2/libremesh/develop-dev-brunch/lime-sdk/output/ar71xx/generic/Default/lime_default/ ) 

Both seem to return the same compile error below:

 ./cooker -b ramips/mt7620 creates the following compile error.

package/Makefile:105: recipe for target '/home/tiger/lime-sdk/snapshots/ramips/mt7620/sdk/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/tiger/lime-sdk/snapshots/ramips/mt7620/sdk/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tiger/lime-sdk/snapshots/ramips/mt7620/sdk'
/home/tiger/lime-sdk/snapshots/ramips/mt7620/sdk/include/toplevel.mk:200: recipe for target 'world' failed
make: *** [world] Error 2
make: Leaving directory '/home/tiger/lime-sdk/snapshots/ramips/mt7620/sdk'

ilario, 2017-10-30:
Hi @AndroidCloned, can you post the full output of cooker (or a link to a pastebin with it)?
Likely real the error is hidden in some row above what you posted here.
Thanks!

AndroidCloned, 2017-10-30:
@ilario Thanks for the quick response! Much appreciated. Here is the paste bin on a fresh git clone  using the develop branch. https://pastebin.com/9g3gHmbQ

ilario, 2017-10-30:
Never saw that error...
Can you try if using lime-sdk master branch you get the same error?

AndroidCloned, 2017-10-30:
If I use the master branch I can compile as normal. Unfortunately, I have several archer-c59-v1 routers that I am working with which are not in the master branch. @FreifunkUFO is the only one who has a working libremesh build from April.

ilario, 2017-10-30:
I reproduced obtaining exactly the same error. @p4u pls

ilario, 2017-10-30:
Pau had your same error months ago: https://github.com/libremesh/lime-sdk/issues/2#issuecomment-304510970

ilario, 2017-10-31:
It looks like a problem from LEDE to me.

Anyway the problem is that two files `lede/ramips/mt7620/sdk/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/compat-wireless-2017-01-31/include/linux/ath9k_platform.h` and `lede/ramips/mt7620/sdk/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-4.9.58/include/linux/ath9k_platform.h` are different, the first has a line more than the second:
```
const char *led_name;
```

The horrible but working workaround for me was to copy the first on the second.

```
cd lime-sdk
git checkout develop
cp lede/ramips/mt7620/sdk/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/compat-wireless-2017-01-31/include/linux/ath9k_platform.h lede/ramips/mt7620/sdk/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-4.9.58/include/linux/ath9k_platform.h
./cooker -b ramips/mt7620
```

FreifunkUFO, 2017-10-31:
i think it a problem with lime-sdk development brunch.
i dont understand these settings there, it seems to be a wild mix. https://github.com/libremesh/lime-sdk/compare/develop but its not lede-trunk in there. maybe its also affected from that "frozen wireless-compat"?

you can see it in your pastebin log file, "compat-wireless-2017-01-31", but on lede-trunk there is a newer version (f.i. krack issue).

p.s. please try target tplink 4300. it may also fail. lets discuss it on #2

ilario, 2017-10-31:
@FreifunkUFO I also suspected about the frozen compat-wireless but I could find no references to it anywhere in the code so I suppose it's not used (still waiting for @p4u's answer on libremesh/lime-packages#144).

I tried tplink 4300, had the same problem and solved with the identical workaround. I'm going to comment on #2 and close this if you agree.

AndroidCloned, 2017-10-31:
I believe this is the case as well. I was comparing sources and came across https://github.com/lede-project/source/blob/master/package/kernel/mac80211/Makefile as well as the kernel source which is 13 days old https://git.kernel.org/pub/scm/linux/kernel/git/wireless/wireless-testing.git

Thank you everyone for resolving my issue. This does mean our snapshot builds are vulnerable to the krack attacks and bufferbloat that was recently fixed.

ilario, 2017-10-31:
Ok, let's move the discussion on #2 and on #33 for having a branch for 17.01.4 (which is be Krack safe I suppose...?)

closed 2017-10-31 by ilario