libremesh/lime-sdk #44: Branch for compiling with new web UI

ilario, 2017-11-28:
The new web interface developed by @gmarcos87 is based on LuCI2 and is located here:
https://github.com/libremesh/lime-packages-ui
and here:
https://github.com/libremesh/lime-app/

We need a branch for compiling an updated 17.01.4 LEDE with the new UI, it will be something like this old branch:
https://github.com/libremesh/lime-sdk/compare/luci2

Some notes:

* abandoned https://github.com/libremesh/lime-ui-ng has to be removed from feeds.conf.default
* new https://github.com/libremesh/lime-packages-ui has to be added to feeds.conf.default
* a new flavor has to be added, I don't know if the legacy webui can be there or should be excluded

nicopace, 2017-11-28:
It is not based on LuCI2 ;) It is based on his own doing (using Preact and PWA)

ilario, 2018-02-04:
Updates on this? @gmarcos87 @nicopace 
Do you think that the new webui will be default on upcoming release?

nicopace, 2018-02-04:
Have already added it to lime_default profile!
You can close this

ilario, 2018-02-04:
EPIC!

closed 2018-02-04 by ilario