libremesh/lime-sdk #49: Migration to OpenWrt

ilario, 2018-01-08:
Mostly solved by #48 
LEDE is still mentioned in this repository here:
https://github.com/libremesh/lime-sdk/search?q=lede

aparcar, 2018-05-30:
@patogit are you up for that job?

nordurljosahvida, 2018-11-13:
Hi. I'm not sure what the intended behaviour is, but my images build with lime-sdk are still based on LEDE 17.01, why are they not using openwrt?

Thanks!

ilario, 2018-11-13:
@nordurljosahvida upcoming release will be based on OpenWrt, but current stable release (Dayboot Rely, 17.06) is based on LEDE

nordurljosahvida, 2018-11-13:
@ilario thanks for the info

ilario, 2018-11-23:
I don't know, let's ask @p4u @aparcar @nicopace @altergui

nordurljosahvida, 2018-11-25:
@ilario These routers are amazing by the way. They might not be libre as the LibreRouter but they're 30€ each, AC1200, very beautiful and small, and seem to work great with openwrt. It takes about 15 minutes to [flash openwrt](https://forum.openwrt.org/t/xiaomi-wifi-router-3g/5377/57) onto them as you need to enable SSH onto them first, but once that's done you're good to go. I think they'd be an amazing device for LiMe networks, especially given their excellent quality and price. I hope we can get a build for them soon! Thanks again for the help!

ilario, 2018-11-25:
while we wait for @p4u's answer, you can try to compile using lime-sdk branch develop, maybe taking some changes from openwrt-18.06 branch for compiling on 18.06 rather than on trunk:

https://github.com/libremesh/lime-sdk/compare/develop...openwrt-18.06

nordurljosahvida, 2018-11-25:
Perfect thanks! I've used the 18.06 branch as it is for now, to test things out, and I see the profile!

```
user@ubuntu:~/lime-sdk$ ./cooker --profiles=ramips/mt7621 | grep 3g
mikrotik_rbm33g:
mir3g:
```

nordurljosahvida, 2018-11-27:
@ilario i'm having some issues on 18.06 on develop branch. when lime-config runs, it breaks meshing with other nodes. so if i just flash a node and start it up, it works. if i change the node name, either from web or from `/etc/config/lime`, then run `lime-config`, things break. resetting to defaults from backup/flash firmware page works. should i open an issue or would it just generate noise, being for 18.06? thanks again

nordurljosahvida, 2018-11-28:
ok great thanks

ilario, 2019-01-18:
Some more has been addressed in #97

closed 2019-06-26 by p4u

ilario, 2019-06-26:
@p4u are you sure it is fixed?
There are still mentions to LEDE in Readme and in flavors.conf:
https://github.com/libremesh/lime-sdk/search?q=lede

patogit, 2019-06-28:
current LibreMesh development by the LibreRouter team is happening [at this repo](https://github.com/LibreRouterOrg/openwrt/tree/librerouter-18.06) and is not using the lime-sdk, but instead the process described in the readme there.

p4u, 2019-06-28:
> @p4u are you sure it is fixed?
> There are still mentions to LEDE in Readme and in flavors.conf:
> https://github.com/libremesh/lime-sdk/search?q=lede

Fixed!

> current LibreMesh development by the LibreRouter team is happening [at this repo](https://github.com/LibreRouterOrg/openwrt/tree/librerouter-18.06) and is not using the lime-sdk, but instead the process described in the readme there.

LibreMesh and LibreRouter are diferent projects. LibreMesh should be compatible with LibreRouter but also with many other devices. It makes sense that LibreRouter has its own specific tools, but LibreMesh should keep its generic behaviour (not dependent on any specific hardware device).

Cheers.

patogit, 2019-06-28:
> LibreMesh should be compatible with LibreRouter but also with many other devices. It makes sense that LibreRouter has its own specific tools, but LibreMesh should keep its generic behaviour (not dependent on any specific hardware device).

I agree!!! I don't understand the function of lime-sdk given that now I'm using the OpenWrt method of making firmware (git clone, update feeds, install feeds, edit .config, make), which LibreRouter development uses too.