libremesh/lime-sdk #50: Add support for feeds.conf src options

nicopace, 2018-01-08:
In https://wiki.openwrt.org/doc/devel/feeds openwrt sdk and ib support multiple feed fetching methods.
Giving support to at list src-link or src-spy will allow to have local repositories integrated within lime-sdk.

p4u, 2018-01-09:
This is already possible be adding your own feeds in feeds.conf.default

There is a documentation about how to use your own feed in the README file of lime-sdk

nicopace, 2018-01-09:
I tried it a few days ago and src-link and src-cpy where not working.
When I changed them to src-git it started working.

Could it be because it is not implemented here?
https://github.com/libremesh/lime-sdk/blob/master/cooker#L280

p4u, 2018-01-09:
Ok, now I understand the issue.

And yes it probably need some changes of the code. However you can
always use git in a local directory

Example of what I'm saying:

`mkdir feed1
cd feed1
git init
cd ..
git clone ./feed1 feed2`

At this point feed2 is a clone of feed1 and the origin is a local
directory file.

nicopace, 2018-01-09:
Yes, I was doing that... just wanted to state that it in order to comply with how openwrt defines the feeds.conf, there was some lacking. Not difficult dough'... could be left as an exercise for someone starting with lime-sdk :)

p4u, 2018-01-12:
Ok, agreed.

Right now cooker ignores the non-git feeds 
https://github.com/libremesh/lime-sdk/blob/master/cooker#L280

That may be changed to support also more kind of feeds.

aparcar, 2018-05-30:
@nicopace please review PR

aparcar, 2018-06-13:
fixed

commit 73360100 (2018-01-17): fix #50
commit deb5c89e (2018-01-17): Merge pull request #51 from aparcar/patch-2

fix #50
commit 01d9cf41 (2018-01-17): fix #50
commit 1d307b01 (2018-06-05): Merge pull request #69 from aparcar/src-link-cpy

handle src-link and src-cpy. fix #50
closed 2018-06-13 by aparcar