libremesh/lime-sdk #85: Install OpenWRT kmod-usb-serial-ftdi package, tmux, screen

1am, 2018-07-05:
Hello,

I'm trying to add few standard packages from OpenWRT package repository and I see that they are being downloaded into packages directory but when I try running cooker with modification to shoe menuconfig instead of defconfig I can't select or find them. 
What is the want to add the following:
* kmod-usb-serial-ftdi
* tmux
* screen

I don't get the fill cooker flow yet, could someone please help me out with it? I've added my own custom repository and packages without problems but these are proving to be more difficult.

aparcar, 2018-07-05:
how do you add them? You could for instance extend the `flavors.conf` file

1am, 2018-07-05:
I've added my own flavor in **flavors.conf.local**

```
lime_my_flavor="lime-full lime-app MyApp my-app-2 screen tmux kmod-usb-serial-ftdi -dnsmasq -libudev-fbsd"
```

Afterwards I build everything by:

```
COOKER_TARGET=ar71xx/generic
COOKER_PROFILE=carambola2
COOKER_FLAVOR=lime_my_flavor

# Optionally when feeds need update
# ./cooker -f --force

./cooker \
  -b ${COOKER_TARGET}

./cooker \
  -c ${COOKER_TARGET} \
  --profile=${COOKER_PROFILE} \
  --flavor=${COOKER_FLAVOR}
```

"MyApp" and "my-app-2" from custom repository (added proper entry to **feeds.conf.default.local**) are visible, others aren't. Also resulting build contains my apps but not the other, standard packages.

1am, 2018-07-05:
Maybe there is some blacklist for modules? The thing I can't find is where these modules are removed as clearly they're downloaded to local feeds/packages folder.

aparcar, 2018-07-06:
I think `feeds.conf.default.local` is nowhere checked in the `cooker` script, did you changed it in the options file?

1am, 2018-07-06:
It works. I think this part checks for **feeds.cond.default.local** https://github.com/libremesh/lime-sdk/blob/master/cooker#L267 I've followed https://github.com/libremesh/lime-sdk#add-your-own-feed-repository when getting started.

My **feeds.cond.default.local** is a copy of https://github.com/libremesh/lime-sdk/blob/master/feeds.conf.default with one additional line at the end which points to my repo.
And the thing is my repo works well while packages from https://github.com/libremesh/lime-sdk/blob/master/feeds.conf.default#L2 are not selectable, like https://git.openwrt.org/?p=feed/packages.git;a=tree;f=utils/tmux;h=3ea84b4b84267b57420aa2a070228e03191214d7;hb=HEAD for instance.

aparcar, 2018-07-11:
> I think feeds.conf.default.local is nowhere checked in the cooker script, did you changed it in the options file?

Sorry this was completely wrong, I meant to write `flavors.conf.local`, which isn't even related but I initially misread your questions.

So you add packages to flavor but don't add the `--remote` parameter, thereby the packages are searched locally. However, the `cooker -b` doesn't know you have  added special packages via flavors and rely on the standard config in [libremesh.sdk.config](https://github.com/libremesh/lime-sdk/blob/master/libremesh.sdk.config).

You can manually add your preferred packages by adding lines like `CONFIG_PACKAGE_tmux=m`, however that's no fun so I fixed that in the [development branch](https://github.com/libremesh/lime-sdk/commit/50c681c780033e55795967fe143c13ec09b1fe93), feel free to try that. You can edit in [options.conf](https://github.com/libremesh/lime-sdk/blob/develop/options.conf#L25) extra packages you want to compile.

Please test the development branch if it works for you, would be happy to get some feedback.

1am, 2018-07-12:
Hi, 

Thank you for your reply. 

I've tried working on cooker checked out from [development branch](https://github.com/libremesh/lime-sdk/commit/50c681c780033e55795967fe143c13ec09b1fe93) and adapted my build script along with the options. The result is that it's now failing on my custom packages again so probably missed something? Didn't yet dig through the changes between development and master branches

### My build script

-----

```
./cooker -f --force

./cooker \
  -b ${COOKER_TARGET} \
  --remote

./cooker \
  -c ${COOKER_TARGET} \
  --profile=${COOKER_PROFILE} \
  --flavor=${COOKER_FLAVOR} \
  --remote

```

### My options.conf.local

Changed `sdk_install_packages` to `sdk_compile_packages`

```
sdk_compile_packages="libustream-openssl firewall screen tmux kmod-usb-serial-ftdi MyApp my-app-2"
```

1am, 2018-07-30:
@aparcar Do you maybe have a hint on what could be wrong with my setup?

1am, 2018-08-03:
What eventually seemed to do the job is the following flags

```
/cooker \
  -b ${COOKER_TARGET} \
  --force-local

./cooker \
  -c ${COOKER_TARGET} \
  --profile=${COOKER_PROFILE} \
  --flavor=${COOKER_FLAVOR} \
  --extra-pkg=libustream-openssl firewall screen tmux kmod-usb-serial-ftdi MyApp my-app-2
```

I'm not sure if it was `--force-local` or specyfying the `--extra-pkg` list but the good thing is that after quick testing it seemed to inclide all the required packages! :)

closed 2018-08-03 by 1am

aparcar, 2018-08-03:
Glad it's fixed, sorry for not responding I have currently no build environment to test it meyself