libremesh/lime-sdk #86: lime-smart-wifi

netzcoop, 2018-08-04:
Hello, 
how can we add lime-smart-wifi to the packagebuild.

we tried: 
in flavors.local:
lime_camp="lime-basic-no-ui lime-debug  -dnsmasq -wpad-mini wpad-mesh lime-smart-wifi"

and in the 
feeds.config:
src-git libremesh https://github.com/libremesh/lime-packages.git

The Cooker comand is:
 ./cooker -c ar71xx/generic --flavor=lime_camp --profile=tl-wdr4300-v1 --community=camp/cpublic   ( --extra-pkg="lime-smart-wifi")
with and without the --extra-pkg="lime-smart-wifi" option.

Collected errors:
 * opkg_install_cmd: Cannot install package lime-smart-wifi.

what we do wrong?

thank you!

aparcar, 2018-08-05:
The package isn't (yet) compiled for libremesh. However I started the compile process for both 17.06 and snapshots, within the next ours it should be possible to install.

To build it on your own you have to edit `libremesh.sdk.config` and add `CONFIG_PACKAGE_lime-smart-wifi=m` and run `./cooker -b ar71xx/generic`. This does take some time...

netzcoop, 2018-08-07:
Nice it is working (compiled with 17.06) with  our wdr4300 and ubnt-unifiac-lite.
With adding to lime_defaults:
config smart_wifi 'smart_wifi'
    option ap_2ghz '1'
    option ap_5ghz '0'
    option mesh_2ghz '0'
    option mesh_5ghz '1'
    option all_to_mesh '0'
    option all_to_ap '0'
    list channels_2ghz '11 6 1'
    list channels_5ghz '44 128 100'

dangowrt, 2018-08-07:
On Tue, Aug 07, 2018 at 02:04:31AM -0700, netzcoop wrote:
> Nice it is working (compiled with 17.06) with  our wdr4300 and ubnt-unifiac-lite.
> With adding to lime_defaults:

Nice! So that's where the wifi settings (missing in your other post)
come from in the end.

> config smart_wifi 'smart_wifi'
>     option ap_2ghz '1'
>     option ap_5ghz '0'
>     option mesh_2ghz '0'
>     option mesh_5ghz '1'
>     option all_to_mesh '0'
>     option all_to_ap '0'
>     list channels_2ghz '11 6 1'

PLEASE use '1 5 9 13' instead, this setting wastes about 1/3 of the
spectrum only to support ancient CCK rates (ie. we can have one
additional non-overlapping channel in 2.4GHz if we just set
option legacy_rates '0')

>     list channels_5ghz '44 128 100'

netzcoop, 2018-08-07:
thanx for the hint i will fix this.

aparcar, 2018-08-07:
Please create a PR to change the defaults, I'm happy to merge it

netzcoop, 2018-08-07:
@aparcar:  sry do you mean dangowrt with the list channels or me with change: option mesh_2ghz '0' in the 90_lime-smart-wifi ?

ilario, 2018-10-01:
@netzcoop I suppose the change should be in [90_lime-smart-wifi](https://github.com/libremesh/lime-packages/blob/master/packages/lime-smart-wifi/files/etc/uci-defaults/90_lime-smart-wifi) from:

` uci add_list lime-defaults.smart_wifi.channels_2ghz="1 11 6"`

to

` uci add_list lime-defaults.smart_wifi.channels_2ghz="1 5 9 13"`

Can anyone do the pull request?

nicopace, 2018-10-07:
It is 1, 6 and 11 @ilario 
https://en.wikipedia.org/wiki/List_of_WLAN_channels

closed 2018-10-07 by nicopace

nicopace, 2018-10-07:
Have not seen @dangowrt comment...
Can you share any reference to leave as documentation for this change?
Couldn't find any around... found mentions of it here:
* https://www.metageek.com/training/resources/design-dual-band-wifi.html
* European nations commonly use channels 1, 5, 9, and 13, four channels with some overlap: https://cromwell-intl.com/networking/wlan-specs.html
* AeroHive says: In Europe, a 2.4 GHz four-channel reuse pattern of channels 1, 5, 9 and 13 is sometimes deployed as shown in Figure 6. Although there is a small amount of frequency overlap between those four channels, the performance might in some cases be better if the medium contention overhead of co-channel cooperation can be reduced because there is less channel bleed over: https://blog.aerohive.com/2-4-ghz-channel-planning/

What does it mean "some overlap if using 1,5,9,13"? how is it going to affect?