libremesh/lime-sdk #87: Problem with configuration of single radio TP-Link CP510 with encrypted mesh

netzcoop, 2018-08-07:
1) The TP-Link CP510 (having only 5G) creates a mesh over the 5G (encrypted) and we can see him with bmx6 (orin the other setup with bmx7), but he is not in the configured subnet (10.5*.*) instead he uses the ip 192.168.1.1.
So it is not possible to use the router as Gateway.

We want to use this router as a bridge for longer distances and to offer net over wdr4300 or the TP-Link CP210.

We tried with bmx6 and bmx7 

Packages used:
* lime-basic-no-ui hostapd-mini lime-debug  -dnsmasq -wpad-mini wpad-mesh
(For the bmx7 setup: bmx7 additional: lime-proto-bmx7 bmx7 bmx7-auto-gw-mode)

Our configuration lime_defaults:
 config lime system
	option hostname '%M4%M5%M6'
	option domain 'mesh.camp.local'
	option keep_on_upgrade 'libremesh base-files-essential /etc/sysupgrade.conf'

config lime network
	option primary_interface eth0
	option main_ipv4_address '10.5.0.0/21'
	option main_ipv6_address '2801:01e8:2::/64'
	option bmx6_mtu '1500'
	list protocols ieee80211s
	list protocols lan
	list protocols anygw
	list protocols batadv:%N1
	list protocols bmx6:13
	list protocols olsr:14
	list protocols olsr6:15
	list protocols olsr2:16
    list resolvers 8.8.8.8  
	list resolvers 4.2.2.2   # b.resolvers.Level3.net
	list resolvers 141.1.1.1 # cns1.cw.net
	list resolvers 2001:470:20::2 # ordns.he.net
	option bmx6_over_batman false
	option bmx6_pref_gw none
	option bmx6_publish_ownip true
	option anygw_mac "aa:aa:aa:%N1:%N2:aa"

config lime wifi
    option channel_2ghz '1'
    list modes 'ap'
    list modes 'ieee80211s'
    option ap_ssid 'CAMP'
    option ap_encryption 'psk2+ccmp'
    option ap_key 'PASSWORD'

Do we need a special Package config to get the Setup working with Hardware with Single Radio?

dangowrt, 2018-08-07:
Hi!

On Tue, Aug 07, 2018 at 02:26:58AM -0700, netzcoop wrote:
> 1) The TP-Link CP510 (having only 5G) creates a mesh over the 5G (encrypted) and we can see him with bmx6 (orin the other setup with bmx7), but he is not in the configured subnet (10.5*.*) instead [...]

'bridging' is a layer-2 concept. you can do that with batman-adv, but
what bmx6/7 does is routing, ie. setup routes between layer-3 subnets.
Please elaborate a bit more **which** router cannot be used as a
gateway into **which** network and why.

> We tried with bmx6 and bmx7
>
> Packages used:
> * lime-basic-no-ui hostapd-mini lime-debug  -dnsmasq -wpad-mini wpad-mesh
> (For the bmx7 setup: bmx7 additional: lime-proto-bmx7 bmx7 bmx7-auto-gw-mode) [...]

you miss all the mesh-specific wifi settings in the wifi section.
maybe add something like:
    option htmode_5ghz 'HT40'
    option channel_5ghz '36'
    option country 'NL'
    option ieee80211s_mesh_id 'LiMe'
    option ieee80211s_mesh_fwding '0'
    option ieee80211s_encryption 'psk2'
    option ieee80211s_key 'MESHPASSWORD'

Once you got that, call lime-config and see if it complains about
anything. Wrongly set network stuff is often caused by lime-config
falling apart due to missing settings in /etc/config/lime or
/etc/config/lime-default

Cheers

Daniel

netzcoop, 2018-08-07:
my bad you are right, i copied the wrong "config lime wifi" we (have multiple tried) use:
config lime wifi
        option channel_5ghz '36'
        option htmode_5ghz 'HT40'
        list modes 'ieee80211s'
        option ieee80211s_mesh_fwding '0'
        option ieee80211s_mesh_id 'camp'
        option ieee80211s_encryption 'psk2+ccmp'
        option ieee80211s_key 'xxxxxxxxxxxxxxxxx'
        option ieee80211s_bssid '15:26:F5:16:44:15'
        option ieee80211s_mcast_rate_5ghz '6000'

With the CPE510 we don't use lime-smart-wifi, as i understand  it should not be needed if we have a single radio (5G)
is the country option obligatory?

And for clarification for readers of this topic the setup is working for:
tl-wdr4300-v1  ubnt-unifiac-lite with the following packages "lime-basic-no-ui hostapd-mini lime-debug  -dnsmasq -wpad-mini lime-smart-wifi wpad-mesh"
And the Full lime-defaults (for 2 2.4G and 5G Radios) config:

config lime system
	option hostname '%M4%M5%M6'
	option domain 'mesh.camp.local'
	option keep_on_upgrade 'libremesh base-files-essential /etc/sysupgrade.conf'
config lime network
	option primary_interface eth0
	option main_ipv4_address '10.5.0.0/21'
	option main_ipv6_address '2801:01e8:2::/64'
	option bmx6_mtu '1500'
	list protocols ieee80211s
	list protocols lan
	list protocols anygw
	list protocols batadv:%N1
	list protocols bmx6:13
	list protocols olsr:14
	list protocols olsr6:15
	list protocols olsr2:16
    list resolvers 8.8.8.8  
	list resolvers 4.2.2.2   # b.resolvers.Level3.net
	list resolvers 141.1.1.1 # cns1.cw.net
	list resolvers 2001:470:20::2 # ordns.he.net
	option bmx6_over_batman false
	option bmx6_pref_gw none
	option bmx6_publish_ownip true
	option anygw_mac "aa:aa:aa:%N1:%N2:aa"

config lime wifi
        option channel_2ghz '6'
        option channel_5ghz '44'
        option htmode_5ghz 'HT40'
        list modes 'ap'
        list modes 'ieee80211s'
        option ap_ssid 'CAMP'
        option ap_encryption 'psk2+ccmp'
         option ap_key 'PASSWORD'
        option ieee80211s_mesh_fwding '0'
        option ieee80211s_mesh_id 'MESH'
        option ieee80211s_encryption 'psk2+ccmp'
        option ieee80211s_key 'PASSWORD'
        option ieee80211s_bssid '16:27:F6:27:57:14'
        option ieee80211s_mcast_rate_5ghz '6000'
config smart_wifi 'smart_wifi'
    option ap_2ghz '1'
    option ap_5ghz '0'
    option mesh_2ghz '0'
    option mesh_5ghz '1'
    option all_to_mesh '0'
    option all_to_ap '0'
    list channels_2ghz '11 6 1'
    list channels_5ghz '44 128 100' 

As in the other topic sayd i will fix the channels to: 1 5 9 13

the tip with lime-config is good now I can see the errors by myself (the problem is that we don't use the routers all time of the year so we are not in it so deeply ):
/usr/lib/lua/lime/utils.lua:68: attempt to concatenate local 'input' (a nil value)
stack traceback:
        /usr/lib/lua/lime/network.lua:278: in function </usr/lib/lua/lime/network.lua:278>
        /usr/lib/lua/lime/utils.lua:68: in function </usr/lib/lua/lime/utils.lua:63>
        (tail call): ?
        /usr/lib/lua/lime/utils.lua:93: in function 'applyNetTemplate10'
        /usr/lib/lua/lime/proto/batadv.lua:72: in function 'setup_interface'
        /usr/lib/lua/lime/network.lua:277: in function </usr/lib/lua/lime/network.lua:277>
        [C]: in function 'xpcall'
        /usr/lib/lua/lime/network.lua:277: in function </usr/lib/lua/lime/network.lua:236>
        [C]: in function 'xpcall'
        /usr/bin/lime-config:49: in function 'main'
        /usr/bin/lime-config:75: in main chunk
        [C]: ?
Use fallback value for network.bmx7_wifi_rate_max: auto
WARNING: Attempt to access undeclared default for: wifi.ap_ssid
stack traceback:
        /usr/lib/lua/lime/config.lua:41: in function 'get'
        /usr/lib/lua/lime/utils.lua:82: in function 'network_id'
        /usr/lib/lua/lime/utils.lua:93: in function 'applyNetTemplate10'
        /usr/lib/lua/lime/proto/batadv.lua:72: in function 'setup_interface'
        /usr/lib/lua/lime/network.lua:277: in function </usr/lib/lua/lime/network.lua:277>
        [C]: in function 'xpcall'
        /usr/lib/lua/lime/network.lua:277: in function </usr/lib/lua/lime/network.lua:236>
        [C]: in function 'xpcall'
        /usr/bin/lime-config:49: in function 'main'
        /usr/bin/lime-config:75: in main chunk
        [C]: ?

   option channel_5ghz '36'
        option htmode_5ghz 'HT40'
        list modes 'ieee80211s'
        option ieee80211s_mesh_fwding '0'
        option ieee80211s_mesh_id 'camp'
        option ieee80211s_encryption 'psk2+ccmp'
        option ieee80211s_key 'xxxxxxxxxxxxxxxxx'
        option ieee80211s_bssid '15:26:F5:16:44:15'
        option ieee80211s_mcast_rate_5ghz '6000'
And I added to the config:
        list modes 'ap'
        option ap_ssid 'CAMP'
        option ap_encryption 'psk2+ccmp'
        option ap_key 'PASSWORD'

no it is working! thanx a lot!
should it not be possible to configure a device just for mesh without ap? with the encrypted setting the ap must be dissabled on a single  radio router it is only possible to use mesh or ap....

netzcoop, 2018-08-07:
i tested with the bmx7 setup and bmx6 setup...

dangowrt, 2018-08-07:
> should it not be possible to configure a device just for mesh without ap? with the encrypted setting the ap must be dissabled on a single  radio router it is only possible to use mesh or ap....

You can totally do that, however, a lot of values (like addresses) are
calculated on the base of the ap_ssid, so you still need to set that
even if you don't actually use the device as an AP.

netzcoop, 2018-08-07:
ok i understand,
thx for the info

netzcoop, 2018-08-08:
can you tell me what is the preferred/best way to disable the AP if not needed.

The problem (with TP-Link CP510) is that the wlan0-mesh changes from Mode: Mesh Point to Mode: Client even if the AP is disabled....

aparcar, 2018-08-09:
Remove ap and ap-name from the wifi modes section in the lime config file.

ilario, 2018-10-01:
@netzcoop can we close this?

netzcoop, 2018-10-10:
yes it can be closed

closed 2018-10-10 by netzcoop