libremesh/lime-sdk #92: How to add a new feed to a working lime-sdk

gmarcos87, 2018-10-23:
If I have a lime-sdk in use, already with its feeds downloaded and updated and I want to add a new one it is not enough to add it in feed.conf and run --update-feeds.
For lime-sdk to take the new packages I must force the feeds with --force. This results in you having to download all the feeds again, and in contexts like mine of low connectivity, it is often impossible.
Is that normal behavior? Did anyone else get sick to this situation? I must do something different or there is a workaround?

gmarcos87, 2018-10-23:
What I do now is manually enter the sdk folder and run:
`scripts/feeds install -p new_repo_folder_name_in_sdk_feeds -a`

nicopace, 2018-10-23:
I went through that a few times already... but don't remember how it's done now... Will come back to this later.
For me it was something like:
* adding the repo in the libremesh.feeds.conf file (or something like that)
* adding the packages I want to get compiled in a file also in the root of lime-sdk
* another step that I can't recall.

It would be great if this could be simplified... even it could be a command within cooker.