libremesh/lime-sdk #101: Replace lime-full metapackage with a list of packages

ilario, 2019-08-12:
`lime-full` metapackage has been removed months ago (https://github.com/libremesh/lime-packages/pull/513) but it is still included in the lime_default [flavor](https://github.com/libremesh/lime-sdk/blob/develop/flavors.conf).
It has to be replaced with `lime-basic` + a list of packages.

ilario, 2019-08-14:
Seems that the "master" branch I fixed in #99 has been renamed to old_master ( ***WTF*** ) and the current master was renamed from another one. So that this problem is not fixed neither in the current master.

p4u, 2019-08-14:
master lime-sdk branch was pointing to the old and outdated libremesh release which is not maintained . So I decided to remove the support to that release and support only the current master libremesh branch (which contains the last libremesh features). 

If you want to maintain the support to the old libremesh release, let's do it in a branch named "release17.06". But from now on lime-sdk master branch will point to libremesh master branch.

Regarding the issue with lime-full, you'r right. It must be fixed in the new master branch.

ilario, 2019-09-06:
This was fixed in 5d7d1bde6cf53f0030ff50a0a4e0496c4510758a

closed 2019-09-06 by ilario