libremesh/lime-sdk #102: lime-packages/develop branch non-existent

ilario, 2019-08-12:
Currently lime-sdk default branch (which is develop) [is looking for the LibreMesh code on the lime-packages/develop branch](https://github.com/libremesh/lime-sdk/blob/develop/feeds.conf.github#L6) that has been deleted long time ago.
We could point to lime-packages/master branch but, considering that lime-sdk is getting less and less supported, I would keep this tool available just for compiling LibreMesh 17.06, so it should point to lime-packages/17.06 (as it currently happens for lime-sdk/master branch).
Or should we just discontinue it?
@p4u

p4u, 2019-08-14:
Fixed

My approach is to support only libremesh master branch and not the release 17.06 anymore. If you wish to maintain it, lets create a specific branch for this release.

Thanks

ilario, 2019-09-06:
So, it got fixed changing the default branch from develop (still broken) to master (points to the correct branch). Closing.

ilario, 2019-09-06:
LOL no, I was wrong. This is not fixed in neither of the branches.