libremesh/lime-sdk #104: error compiling lime_bmx7 flavor with 18.06.4

pedro-nonfree, 2019-08-15:
Hi,

I have an environment to compile openwrt (so I hope all dependencie are OK). I'm experiencing problems running lime-sdk master branch latest commit https://github.com/libremesh/lime-sdk/commit/eae984428969d10357dc22515bea51c9f0bbd832 (I don't have previous experience building this for earlier commits)

```
git clone https://github.com/libremesh/lime-sdk.git
cd lime-sdk
./cooker -c x86/64 --profile=Generic --flavor=lime_bmx7
```

part of the output

```
 make[3] -C /root/lime-sdk/feeds/libremesh/packages/ubus-lime-utils compile                                         
 make[3] -C /root/lime-sdk/feeds/libremesh/packages/lime-app clean-build                                                           
 make[3] -C /root/lime-sdk/feeds/libremesh/packages/lime-app compile                                                       
 make[3] -C /root/lime-sdk/feeds/libremesh/packages/lime-docs clean-build                                                       
 make[3] -C /root/lime-sdk/feeds/libremesh/packages/lime-docs compile                                                              
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on    
make: *** [/root/lime-sdk/18.06.4/x86/64/sdk/include/toplevel.mk:211: world] Error 1                                       
make: Leaving directory '/root/lime-sdk/18.06.4/x86/64/sdk'                                                                                                            
-> Error compiling SDK                                                                                                                                                  
cp: cannot stat '18.06.4/x86/64/ib/repositories.sdk.conf': No such file or directory                                             
-> Cooking x86/64/Generic
-> Cooking firmware image
--> Selected extra packages: lime-system lime-proto-wan lime-hwd-openwrt-wan lime-debug smonit lime-proto-bmx7 bmx7-auto-gw-mode luci luci-app-bmx7 bmx7-topology bmx7-j
son -dnsmasq dnsmasq-dhcpv6
make: Entering directory '/root/lime-sdk/18.06.4/x86/64/ib'
make[1]: Entering directory '/root/lime-sdk/18.06.4/x86/64/ib'
make[1]: Leaving directory '/root/lime-sdk/18.06.4/x86/64/ib'
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.

(...)

Installing libsmartcols (2.32-2) to root...
Downloading http://downloads.openwrt.org/releases/18.06.4/packages/x86_64/base/libsmartcols_2.32-2_x86_64.ipk
Installing ppp (2.4.7-12) to root...
Downloading http://downloads.openwrt.org/releases/18.06.4/packages/x86_64/base/ppp_2.4.7-12_x86_64.ipk
Installing kmod-lib-crc-ccitt (4.14.131-1) to root...
DoCollected errors:
 * opkg_install_cmd: Cannot install package lime-system.
 * opkg_install_cmd: Cannot install package lime-proto-wan.
 * opkg_install_cmd: Cannot install package lime-hwd-openwrt-wan.
 * opkg_install_cmd: Cannot install package lime-debug.
 * opkg_install_cmd: Cannot install package smonit.
 * opkg_install_cmd: Cannot install package lime-proto-bmx7.
 * opkg_install_cmd: Cannot install package bmx7-auto-gw-mode.
wnloading file:packages/kmod-lib-crc-ccitt_4.14.131-1_x86_64.ipk
Installing kmod-slhc (4.14.131-1) to root...
Downloading file:packages/kmod-slhc_4.14.131-1_x86_64.ipk
Installing kmod-ppp (4.14.131-1) to root...
Downloading file:packages/kmod-ppp_4.14.131-1_x86_64.ipk
Installing ppp-mod-pppoe (2.4.7-12) to root...
Downloading http://downloads.openwrt.org/releases/18.06.4/packages/x86_64/base/ppp-mod-pppoe_2.4.7-12_x86_64.ipk
Installing kmod-pppox (4.14.131-1) to root...
Downloading file:packages/kmod-pppox_4.14.131-1_x86_64.ipk
Installing kmod-pppoe (4.14.131-1) to root...
Downloading file:packages/kmod-pppoe_4.14.131-1_x86_64.ipk
Installing uci (2019-05-17-f199b961-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.4/packages/x86_64/base/uci_2019-05-17-f199b961-1_x86_64.ipk
Package uclient-fetch (2018-11-24-3ba74ebc-1) installed in root is up to date.
Configuring zlib.
Configuring libmbedtls.
Configuring libnl-tiny.
Configuring libubox.
Configuring libuci.
Configuring libubus.
Configuring libiwinfo.
Configuring bmx7.
Configuring luci-lib-json.
Configuring liblua.
Configuring lua.
Configuring luci-lib-nixio.
Configuring luci-lib-ip.
Configuring libjson-c.
Configuring libblobmsg-json.
Configuring rpcd.
Configuring libubus-lua.
Configuring luci-lib-jsonc.
Configuring liblucihttp.
Configuring liblucihttp-lua.
Configuring luci-base.
Configuring luci-mod-admin-full.
Configuring bmx7-json.
Configuring luci-app-bmx7.
Configuring libpthread.
Configuring librt.
Configuring libuuid.
Configuring libf2fs.
Configuring mkf2fs.
Configuring dnsmasq-dhcpv6.
Configuring libuclient.
Configuring uclient-fetch.
Configuring opkg.
Configuring ubusd.
Configuring ubus.
Configuring busybox.
Configuring kmod-nf-reject6.
Configuring libiwinfo-lua.
Configuring kmod-nf-conntrack.
Configuring kmod-nf-flow.
Configuring kmod-lib-crc-ccitt.
Configuring r8169-firmware.
Configuring luci-theme-bootstrap.
Configuring kmod-slhc.
Configuring kmod-ppp.
Configuring kmod-pppox.
Configuring kmod-pppoe.
Configuring kmod-nf-reject.
Configuring kmod-nf-ipt.
Configuring kmod-ipt-core.
Configuring kmod-ipt-conntrack.
Configuring jshn.
Configuring netifd.
Configuring libjson-script.
Configuring ubox.
Configuring procd.
Configuring jsonfilter.
Configuring usign.
Configuring openwrt-keyring.
Configuring fstools.
Configuring fwtool.
Configuring base-files.
Configuring kmod-nf-nat.
Configuring kmod-input-core.
Configuring libblkid.
Configuring libsmartcols.
Configuring partx-utils.
Configuring kmod-mii.
Configuring kmod-r8169.
Configuring kmod-i2c-core.
Configuring libxtables.
Configuring libip4tc.
Configuring libip6tc.
Configuring kmod-nf-conntrack6.
Configuring kmod-ipt-nat.
Configuring firewall.
Configuring luci-app-firewall.
Configuring libcomerr.
Configuring kmod-nf-ipt6.
Configuring kmod-ip6tables.
Configuring odhcp6c.
Configuring uci.
Configuring dropbear.
Configuring kmod-hwmon-core.
Configuring mtd.
Configuring odhcpd-ipv6only.
Configuring luci-proto-ppp.
Configuring ppp.
Configuring logd.
Configuring libss.
Configuring luci-proto-ipv6.
Configuring iptables.
Configuring libext2fs.
Configuring e2fsprogs.
Configuring kmod-e1000.
Configuring kmod-pps.
Configuring uhttpd.
Configuring ip6tables.
Configuring kmod-ptp.
Configuring rpcd-mod-rrdns.
Configuring luci.
Configuring bmx7-topology.
Configuring kmod-e1000e.
Configuring kmod-button-hotplug.
Configuring kmod-i2c-algo-bit.
Configuring kmod-igb.
Configuring kmod-ipt-offload.
Configuring ppp-mod-pppoe.
make[2]: *** [Makefile:153: package_install] Error 255
make[1]: *** [Makefile:114: _call_image] Error 2
make: *** [Makefile:193: image] Error 2
make: Leaving directory '/root/lime-sdk/18.06.4/x86/64/ib'

-> Firmware for target x86/64, profile PROFILE=Generic and flavor lime_bmx7 cooked!
Find the binaries in /root/lime-sdk/output/x86/64/Generic/lime_bmx7 directory
root@cooker2:~/lime-sdk# ./cooker -c x86/64 --profile=Generic --flavor=lime_bmx7
```

p4u, 2019-08-15:
> commit eae9844

There is some error on the building phase, so when trying to "cook" the
packages it fails because they have not been created previously.

Take a look to the README.md https://github.com/libremesh/lime-sdk

I'd execute just the building step, like this:

`J=1 V=s ./cooker -b x86/64`

Check `./cooker --help` also to know this kind of tricks)

It's probably that your host system is missing some library required for
compiling the packages. You can also try to use the option "--remote" to
download the packages instead of compiling them yourself (hope the repos
are up-to-date and on-line).

Cheers.

p4u, 2019-08-15:
I just tried `./cooker -c x86/64 --profile=Generic --flavor=lime_bmx7` in a new lime-sdk setup and it worked fine. So I confirm that is something from your local computer.

pedro-nonfree, 2019-08-15:
ok sorry for not debugging appropiately (`J=1 V=s ./cooker -b x86/64`), false alarm, `svn not found` [1] and seems I could do some openwrt buildings in that system

    apt install subversion

after that `./cooker -c x86/64 --profile=Generic --flavor=lime_bmx7` worked

thanks

[1]

```
-and -not -name '.pkgdir' | xargs -r rm -rf
make[3]: Leaving directory '/root/lime-sdk/feeds/libremesh/packages/lime-app'
time: package/feeds/libremesh/lime-app/compile#0.16#0.05#0.18
make[3]: Entering directory '/root/lime-sdk/feeds/libremesh/packages/lime-docs'
mkdir -p /root/lime-sdk/18.06.4/x86/64/sdk/dl
SHELL= flock /root/lime-sdk/18.06.4/x86/64/sdk/tmp/.lime-docs-2019-01-27-1548584850.tar.xz.flock -c '    echo "Checking out files from the svn repository..."; mkdir -p\
 /root/lime-sdk/18.06.4/x86/64/sdk/tmp/dl && cd /root/lime-sdk/18.06.4/x86/64/sdk/tmp/dl && rm -rf lime-docs-2019-01-27-1548584850 && [ \! -d lime-docs-2019-01-27-1548\
584850 ] && ( svn help export | grep -q trust-server-cert && svn export --non-interactive --trust-server-cert -rHEAD https://github.com/libremesh/lime-web/trunk/docs l\
ime-docs-2019-01-27-1548584850 || svn export --non-interactive -rHEAD https://github.com/libremesh/lime-web/trunk/docs lime-docs-2019-01-27-1548584850 ) && echo "Packi\
ng checkout..." && export TAR_TIMESTAMP="" &&       tar --numeric-owner --owner=0 --group=0 --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c lime-docs-2019-01\
-27-1548584850 |     xz -zc -7e > /root/lime-sdk/18.06.4/x86/64/sdk/tmp/dl/lime-docs-2019-01-27-1548584850.tar.xz && mv /root/lime-sdk/18.06.4/x86/64/sdk/tmp/dl/lime-d\
ocs-2019-01-27-1548584850.tar.xz /root/lime-sdk/18.06.4/x86/64/sdk/dl/ && rm -rf lime-docs-2019-01-27-1548584850;     '
Checking out files from the svn repository...
/bin/sh: 1: svn: not found
/bin/sh: 1: svn: not found
make[3]: *** [Makefile:87: /root/lime-sdk/18.06.4/x86/64/sdk/dl/lime-docs-2019-01-27-1548584850.tar.xz] Error 127
make[3]: Leaving directory '/root/lime-sdk/feeds/libremesh/packages/lime-docs'
time: package/feeds/libremesh/lime-docs/compile#0.14#0.04#0.15
make[2]: *** [package/Makefile:109: package/feeds/libremesh/lime-docs/compile] Error 2
make[2]: Leaving directory '/root/lime-sdk/18.06.4/x86/64/sdk'
make[1]: *** [package/Makefile:103: /root/lime-sdk/18.06.4/x86/64/sdk/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/lime-sdk/18.06.4/x86/64/sdk'
make: *** [/root/lime-sdk/18.06.4/x86/64/sdk/include/toplevel.mk:211: world] Error 2
make: Leaving directory '/root/lime-sdk/18.06.4/x86/64/sdk'
-> Error compiling SDK
```

closed 2019-08-15 by pedro-nonfree