libremesh/lime-sdk #105: the only flavor working seems to be bmx7

brunovianna, 2019-08-27:
When I cook a bmx7 firmware with this command everything runs fine:

 ./cooker -c ar71xx/generic --profile=tl-wdr3500-v1 --flavor=lime_bmx7 --remote

If I try any other flavor like lime_default, lime_mini etc, it won't create the files. Build log
[last-build.log](https://github.com/libremesh/lime-sdk/files/3547500/last-build.log)
 is attached.

ilario, 2019-09-05:
In the log I can't see anything useful...
Can you run cooker prepending a "V=s" string?
Like:
```
V=s ./cooker -c ar71xx/generic --profile=tl-wdr3500-v1 --flavor=lime_SOMETHING --remote
```

brunovianna, 2019-09-06:
sure, here it is. i also dumped the output to a file so you can see the error.

issue #107 seems to be the same

[output.txt](https://github.com/libremesh/lime-sdk/files/3584286/output.txt)
[last-build.log](https://github.com/libremesh/lime-sdk/files/3584284/last-build.log)

ilario, 2019-09-06:
Ehm, I cannot really understand what's wrong from the log...
@p4u can you have a look?