libremesh/lime-sdk #106: dnsmasq-lease-share conflicts with shared-state-dnsmasq_leases

nybash, 2019-08-28:
I wonder but it is happening!

```
Collected errors:
 * check_data_file_clashes: Package dnsmasq-lease-share wants to install file /home/frost/tmp/lede-firmware/lime-sdk/18.06.4/ar71xx/tiny/ib/build_dir/target-mips_24kc_musl/root-ar71xx/etc/uci-defaults/90_dnsmasq-lease-share
	But that file is already provided by package  * shared-state-dnsmasq_leases
 * check_data_file_clashes: Package dnsmasq-lease-share wants to install file /home/frost/tmp/lede-firmware/lime-sdk/18.06.4/ar71xx/tiny/ib/build_dir/target-mips_24kc_musl/root-ar71xx/usr/bin/dnsmasq-lease-share.sh
	But that file is already provided by package  * shared-state-dnsmasq_leases
 * opkg_install_cmd: Cannot install package lime-basic.
Collected errors:
 * check_data_file_clashes: Package dnsmasq-lease-share wants to install file /home/frost/tmp/lede-firmware/lime-sdk/18.06.4/ar71xx/tiny/ib/build_dir/target-mips_24kc_musl/root-ar71xx/etc/uci-defaults/90_dnsmasq-lease-share
	But that file is already provided by package  * shared-state-dnsmasq_leases
 * check_data_file_clashes: Package dnsmasq-lease-share wants to install file /home/frost/tmp/lede-firmware/lime-sdk/18.06.4/ar71xx/tiny/ib/build_dir/target-mips_24kc_musl/root-ar71xx/usr/bin/dnsmasq-lease-share.sh
	But that file is already provided by package  * shared-state-dnsmasq_leases
 * opkg_install_cmd: Cannot install package lime-basic.

```

[lime_build.txt](https://github.com/libremesh/lime-sdk/files/3548210/lime_build.txt)

ilario, 2019-09-05:
Which lime-sdk branch are you using?

dnsmasq-lease-share pertains to 17.06 release

shared-state-dnsmasq_leases pertains to the yet to be released release, and is the replacement of dnsmasq-lease-share so it makes sense that they cannot be together, right @G10h4ck?

Could you inspect what selects each of these packages?

nybash, 2019-09-07:
Branch is master

ilario, 2019-09-07:
So do you want to compile LibreMesh 17.06 (in which case I should study how to do this with OpenWrt buildroot, easy but I fear undocumented) or the latest LibreMesh code [in this case I suggest you to use the instructions for the buildroot (update under revision [here](https://github.com/libremesh/lime-web/blob/b2faddc205a9f0f631b78d1dc04fc3811360337e/development.txt#L45) and official instructions [here](https://libremesh.org/development.html)) but we can also try to understand what selects both packages in lime-sdk, on this latter point your help would be most welcome]?