libremesh/lime-sdk #108: Project still active? opkg_download URLs offline

nordurljosahvida, 2020-12-05:
After two years I'm trying to build once again for a project, I'm getting:

```
Collected errors:
 * opkg_download: Failed to download http://snapshots.libremesh.org/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://chef.libremesh.org/network-profiles//Packages.gz, wget returned 8.
```

I used to use branch develop, but I see others are having trouble. I've tried with master, same results.

I also get:

```
Build dependency: Please install the GNU C Compiler (gcc) 4.8 or later
Build dependency: Please install the GNU C++ Compiler (g++) 4.8 or later
Build dependency: Please install Python 2.x
```

on Ubuntu 20 although I have the first two installed [for the third instead, is python2 still really required?]

Also, the new chef.libremesh.org doesn't support communities, right?

Thanks!

ilario, 2020-12-05:
Yesss, the project is still active, but currently no SDK nor Chef are supported.

The only currently supported compilation method is the Buildroot one described on this page:

https://libremesh.org/development.html

This week @spiccinini is likely going to tag a release (3 years since the last official one!!) so the instructions currently on the linked page will result in a good and tested firmware.

ilario, 2020-12-05:
For staying updated please join the lime-users mailing list and the Element/IRC chat: https://libremesh.org/communication.html

nordurljosahvida, 2020-12-05:
ok great thanks! I've noticed some noise in the ML but haven't been keeping track all that much. So will SDK/Chef be re-introduced in the future? What's the best way to perform a custom deployment on several routers right now, with AP names and other custom stuff applied?

ilario, 2020-12-05:
If you indeed want to resurrect lime-sdk, seems easy.
The new directions of the repositories are listed here:
https://github.com/libremesh/lime-packages/blob/e78b43a72367508da25f6eb16c410b16328be9d9/packages/lime-system/files/etc/uci-defaults/92_add-lime-repos#L16-L21
And this has to be reflected here:
https://github.com/libremesh/lime-sdk/blob/master/libremesh.repositories.conf

And yes, python2 is required.
I suppose that the lime-sdk will require a subset of the packages required for the buildroot, try installing the packages suggested on this page:
https://libremesh.org/development.html#compiling_libremesh_from_source_code
This means that the dependencies on https://github.com/libremesh/lime-sdk/blob/master/Dockerfile could need updating.

In the options on https://github.com/libremesh/lime-sdk/blob/master/options.conf you'll have to update the OpenWrt version to 19.07.4 or 18.06.9
And also here: https://github.com/libremesh/lime-sdk/blob/master/feeds.conf.default

The flavors described in https://github.com/libremesh/lime-sdk/blob/master/flavors.conf are extremely outdated and have to be updated to the instructions on the development page of the website.

The suggested target for Atheros routers is now ath79 rather than ar71xx.

If you manage to have it working, please consider making a pull request with the fixes!!!!!
Thanks,
Ilario

ilario, 2020-12-05:
> ok great thanks! I've noticed some noise in the ML but haven't been keeping track all that much. So will SDK/Chef be re-introduced in the future? What's the best way to perform a custom deployment on several routers right now, with AP names and other custom stuff applied?

We included the possibility (@aparcar did) to use the network-profiles in the buildroot.
The network-profiles will have to be updated in order to be used as package: https://github.com/libremesh/network-profiles/issues/66

Then you'll find the profiles in the LibreMesh category of the buildroot menuconfig.

ilario, 2020-12-05:
Anyway I'm not sure the lime-sdk is the best choice.
I did like a lot @aparcar's metabuilder: https://github.com/aparcar/openwrt-metabuilder but now it's archived.
@aparcar what would you suggest to adopt?

aparcar, 2020-12-06:
The feeds are outdated, the rest may work. I'm no longer maintaining these scripts.

ilario, 2020-12-06:
Ok :) and do you know if there is any actively maintained alternative? @aparcar 

If no better alternative comes up, we should evaluate whether to update lime-sdk or fork openwrt-metabuilder (I prefer the second option).

nordurljosahvida, 2020-12-06:
What are you doing now to build for your communities?

nordurljosahvida, 2020-12-06:
And among all of these projects, what has had the most success historically, for what reason were they stopped being maintained, and which one would you like to adopt in the future except for buildroot?

ilario, 2020-12-07:
> What are you doing now to build for your communities?

I have no network community right now, neither in Tarragona nor in Verona.
Can you ask this question in the lime-users mailing list where more people can read and answer?

> And among all of these projects, what has had the most success historically,

For LibreMesh we never officially supported openwrt-metabuilder but we did support lime-sdk.

> for what reason were they stopped being maintained,

I don't know. @aparcar @p4u 
The death of lime-sdk is possibly related to the development of LibreRouterOS (a distribution of LibreMesh) which got all the interest of LibreMesh developers and that was being done with the buildroot.
The archiving of openwrt-metabuilder is more or less explained in its readme:
"This code is no longer maintaned (tho likely works). It was initially created to be used by ASU which however switched to a full Python implementation."

> and which one would you like to adopt in the future except for buildroot?

I like openwrt-metabuilder because in one configuration file you can easily tune everything you need (for example see this: https://github.com/aparcar/openwrt-metabuilder/blob/master/example-libremesh_master.sh).